Diaoul/arduino-ESP8266 #15: All read() methods mixing incomming data from multiple connection to one stream

tprochazka, 2015-01-03:
ESP8266 support multiple incoming connection which is recognizable by "id" in "+IPD,id,length" string.
Problem is that main class directly import Stream interface which is designed handle only one stream.

Related discussion on this topic:
https://github.com/Diaoul/arduino-ESP8266/pull/14

Diaoul, 2015-01-03:
You can subclass the ESP8266 class to achieve what you want or create a wrapper or use @lasselukkari's ESP8266Server class.
The ESP8266 class does not aim to abstract this behavior.

tprochazka, 2015-01-03:
I only want to avoid mix two incoming requests and have possibility to reply to booth of them.
This is currently not possible.
But if you are OK with it, I will keep my fork. My changes are small so I think that I'm able to live with this and keep it updated by changes from you.

@lasselukkari's ESP8266Server class has this some problem, it doesn't solve this problem.

Diaoul, 2015-01-03:
Why no add a readPacket method that reads exactly one IPD block?

tprochazka, 2015-01-04:
I already write, that new method like availablePacket() should be enough.
Rename current available to availablePacket() and original available will simply call _serial->available()
So available() will be in sync with data() and for waiting for next block will be used new method.