Diaoul/arduino-ESP8266 #17: More wifi.send() doesn't work properly

tprochazka, 2015-01-04:
I'm trying to send multiple block like this:

```
wifi.send(1, "A");
wifi.send(1, "B");
wifi.send(1, "C");
wifi.send(1, "D");
```

but library send tot the serial port this:
```
AT+CIPSEND=0,1
AAT+CIPSEND=0,1
AT+CIPSEND=0,1
AT+CIPSEND=0,1
D
```

So I receive only A and D on the, sometimes A and C.
I don't figure out why. Do you have some idea?

The same problem if I want to send messages to different channels like

```
wifi.send(0, "A");
wifi.send(1, "B");
wifi.send(2, "C");
wifi.send(3, "D");
```

tprochazka, 2015-01-04:
So I found problem, ESP return "> busy..." instead of ">", so it is necessary to make some delay before sending next message , according the ESP8266 should be 20ms, but it is not enough, also with 90ms I still get busy sometimes, it is not good. Maby would be usefull if would be possible to send data multipletimes in one block. I mean somethig like this

```
prepareToSend(id, length);
sendData(data1);
sendData(data2);
sendData(data3);
```

Because ESP8266 allow to send 2048 bytes in one block, but preprare 2048 bytes in Arduino memory is problematic, better is prepare smaller block and sent it to the ESP8266

Diaoul, 2015-01-05:
Make sure send doesn't return a timeout, if it does, it's normal you get `busy s...` when looking for `>`