Diaoul/arduino-ESP8266 #18: send() method return TIMEOUT status for "SEND OK"

tprochazka, 2015-01-04:
When ESP8266 return "SEND OK" after call in send() method, library will return ESP8266_COMMAND_TIMEOUT instead of ESP8266_COMMAND_OK

Diaoul, 2015-01-05:
That's weird. When SEND OK is received it should be read as `OK\r\n`: https://github.com/Diaoul/arduino-ESP8266/blob/master/ESP8266.cpp#L834

Diaoul, 2015-01-05:
Unless of course it times out.

Diaoul, 2015-01-05:
You can control send and read timeouts with the `setTimeout` method. Please confirm this works.

Diaoul, 2015-01-05:
I use `_timeout` for network functions because those are subject to variation. Other functions have a more or less stable timeout.

closed 2015-01-05 by Diaoul

welcomeRESPONSE, 2015-09-30:
Hi all. I am a little bit stuck and I could REALLY use some help. So my friend and I are trying to send data through a server to a database (ESP8266 + Arduino, Apache Server and Mysql Database). When we send data, it says “send ok” and after that it says connection timeout. On apache it says 408 error. A few times we got 200 4961 (this is what we want) but that only happens when we send GET /HTTP/1.1 and when we send to GET/index.php?img=pngPlugIn HTTP/1.1. We try to send GET /pm/add_message.php?pmu=RAL005-07001-FL02-0030-2L Host:192.168……. but we get 408 320. Any suggestions? Do you know what we might be doing wrong?