Diaoul/arduino-ESP8266 #22: wifi.send(unsigned int id, String) always returns TIMEOUT

madprogrammi, 2015-01-24:
I'm trying to send a String through a multi-connection instance of the ESP8266 class. When I Serial.print(getStatus(send (id, sendable))); it returns "TIMEOUT" every time. The data is never received by the client. After defining the ESP8266_DEBUG macro, the serial output looks like:

    AT+CIPSEND=0,7
    busy
    TIMEOUT

Where the first two lines are from the ESP8266 library, and TIMEOUT is the output of getStatus();. This code is run in a loop once the client sends a handshake, so I know that the link is up. Any ideas?

Diaoul, 2015-01-24:
Use a longer timeout  with setTimeout. It defaults to 1s which may be too short.

madprogrammi, 2015-01-24:
I had mine set at 3s. Does it need to be longer?

taoyuan, 2015-01-25:
Which version of your esp8266 firmware ? We use 0.94.

madprogrammi, 2015-02-01:
No idea - AT+GMR just responds "OK"

taoyuan, 2015-02-02:
@madprogrammi Try [this](https://github.com/espressif/esp8266_at) firmware.

closed 2015-02-10 by Diaoul