Diaoul/arduino-ESP8266 #23: how to a HTTP GET ?

soundreactor, 2015-01-27:
hi

this might be a stupid question. but i am trying to GET the content from a local server and use it in my arduino code.
the page will give me a variable. no html or nothing.

so how do i use the send ?

this is what i have. 

 '  Serial.print("setWifiMode: ");
    Serial.println(getStatus(wifi.setMode(ESP8266_WIFI_BOTH)));

    // joinAP
    Serial.print("joinAP: ");
    Serial.println(getStatus(wifi.joinAP("interpol_surveillance_van", "mypassword")));

    Serial.print("connect: ");
    Serial.println(getStatus(wifi.connect(ESP8266_PROTOCOL_TCP, IPAddress(192,168,1,7), 13568)));

    Serial.print("send: ");
    Serial.println(getStatus(wifi.send("GET /update/?key=dallameat&field1=22")));'

Jinugc, 2015-01-29:
Same issue for me while using the  connect function 
Serial.print("connect: ");
Serial.println(getStatus(wifi.connect(ESP8266_PROTOCOL_TCP, IPAddress(192,168,1,10), 4444)));

connect: TIMEOUT

Diaoul, 2015-02-10:
Use a higher timeout with `setTimeout` and use latest AT firmware. (0.20 or higher)

closed 2015-02-10 by Diaoul