Diaoul/arduino-ESP8266 #26: MQTT & REST support

Jorgen-VikingGod, 2015-05-20:
Are you planning to add MQTT and REST support to your library?

Diaoul, 2015-05-20:
MQTT il already supported with this library: https://github.com/knolleary/pubsubclient
As for REST this is not planned and I think it's another layer of abstraction that would better fit in another library (that uses this one)

Jorgen-VikingGod, 2015-05-22:
Hi Diaoul, did not know this before.
But how can I use the pubsubclient with your ESP8266Client?
Do you have any examples, how to use this?
thx for the support

Diaoul, 2015-05-25:
The ESP8266Client implements the Client interface so you can use it with PubSubClient. Check out PubSubClient documentation about that.

Jorgen-VikingGod, 2015-05-25:
thx. This works great for me.
So the part of MQTT is solved ;)